JoaoLopesF/RemoteDebug #9: unable to connect after quit

bertmelis, 2017-03-01:
I did a small test with a sample sketch:
connected to telnet: OK
quit connection: OK
connected again: OK, small decrease in free heap
quit connection OK
tried to connect again but "connection refused"

I could post my sketch if needed.

bertmelis, 2017-03-01:
Oeps, I didn't change the port on my Putty instance.

But the memory decrease still exists!

JoaoLopesF, 2017-03-02:
Hi,
I saw about Free heap on Arduino ESP8266 on the internet, as in: https://github.com/esp8266/Arduino/issues/1923.
Iggr, who brought the ESP8266 to the Arduino, said:
This memory should be reclaimed after a couple of minutes, ...

And this occurred in my tests like the sample Basic and occurs in seconds ....

Please check this, you can use the m command in telnet to display the memory

Regards

Joao