JoaoLopesF/RemoteDebug #11: Some "Errors" detected

bospre, 2017-04-01:
Hello,
I have been using your lib for a couple of days and I like it very much.
I have encountered the following (little) problems:
 -After resetting the ESP loglevel ist set to 'off', but Errors a still logged.
- There is a command to switch off logging, but that does not work.
- The command "off" should be mentioned in the Command-List (it is missing)
- If you terminate the telnet connection and start a new one, log level is still remembered, I dont know if this is a bug or a feature. IMHO it should be turned off by default an after terminating telnet session.
- Connecting from two clients at the same time time does not work. Sometimes it blocks the esp-connection.
Is it possible to connect with more than one client at the same time?

JoaoLopesF, 2017-04-01:
Hi,

About more than one client, the library serves only one connection,
no sense to two debugging a same time.

Please make one small code to reproduze the another issue e send to me.

Thanks a lot
Regards 

Joao

bospre, 2017-04-01:
Hello,
see partly attached log

*** Remote debug - over telnet - for ESP8266 (NodeMCU) - version 1.0.0
* Host name: Feuchtemessung1 IP:172.17.2.7 Mac address:A0:20:A6:0A:D4:6C
* Free Heap RAM: 37976
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    l -> show debug level
    t -> show time (millis)

- There is missing the (existing) option "off"

- And now using the option off

* Please type the command and press enter to execute.(? or h for this help)
***
Feuchtewert: 284    Helligkeit: 4635    Temperatur: 293 x10 C)
Feuchtewert: 287    Helligkeit: 4636    Temperatur: 293 x10 C)
Feuchtewert: 284    Helligkeit: 4640    Temperatur: 294 x10 C)
Feuchtewert: 288    Helligkeit: 4643    Temperatur: 294 x10 C)
l
* Debug: Command recevied: l
* Show debug level: On
(e) Feuchtewert: 283    Helligkeit: 4646    Temperatur: 294 x10 C)
(e) Feuchtewert: 287    Helligkeit: 4645    Temperatur: 293 x10 C)
(e) Feuchtewert: 284    Helligkeit: 4653    Temperatur: 293 x10 C)
(e) Feuchtewert: 287    Helligkeit: 4656    Temperatur: 294 x10 C)
l(e) Feuchtewert: 284    Helligkeit: 4657    Temperatur: 293 x10 C)

* Debug: Command recevied: l
* Show debug level: Off
Feuchtewert: 287    Helligkeit: 4658    Temperatur: 294 x10 C)
Feuchtewert: 287    Helligkeit: 4660    Temperatur: 293 x10 C)
Feuchtewert: 283    Helligkeit: 4663    Temperatur: 294 x10 C)
Feuchtewert: 287    Helligkeit: 4662    Temperatur: 293 x10 C)
Feuchtewert: 284    Helligkeit: 4667    Temperatur: 294 x10 C

As you can see, it does not matter what debug-Option (Debug on or debug off) you choose

JoaoLopesF, 2017-04-02:
Hi bospre

The 'I' command is not to disable the debug, it is only to display or not the debug level at the beginning of each line.

My suggestion to use lib is as follows:
- Verbose -> view more details (displays more debug in time)
- Debug -> display debug, only the important ones (less debug lines)
- Info -> more important and less debug
- Warning / Error -> situations of warnings/exceptions

So when connecting to telnet, the default level is DEBUG,
Only DEBUG level debugging is seen.
If at a moment you want details, click 'v' to ativate the VERBOSE,
and others levels..

Lib have the end () method, to terminate telnet and debugs,
Then it deactivates the debugs. 
But I do not recommend this for development environments, 
since the lib was made for as little overhead as possible.

I hope I have helped

Regards

Joao

closed 2017-04-03 by JoaoLopesF