JoaoLopesF/RemoteDebug #12: Show debug level doesn't show debug level

thomasvnl, 2017-04-07:
See this piece of command line print I extracted. I "highlighted" some weird behaviour with two asterisk characters around the value.

```l
* Debug: Command recevied: l
* Show debug level: Off
v
* Debug: Command recevied: v
* Debug level setted to Verbose
l
* Debug: Command recevied: l
* Show debug level: **On**
d
* Debug: Command recevied: d
* Debug level setted to Debug
l
* Debug: Command recevied: l
* Show debug level: **Off**
d
* Debug: Command recevied: d
* Debug level setted to Debug
l
* Debug: Command recevied: l
* Show debug level: **On**
```

It looks like the value for show debug level get's toggled, but also isn't showing the correct debug status.

thomasvnl, 2017-04-07:
I seem to have misinterpret the options, I'm very sorry. I shall close this comment immediately 😉

closed 2017-04-07 by thomasvnl

JoaoLopesF, 2017-04-10:
Ok
Regards

Joao