JoaoLopesF/RemoteDebug #16: ESP32

bospre, 2017-09-07:
Hello,
unfortunately the library does not run with the ESP32. Will there be any adjustments?
Boris

JoaoLopesF, 2017-09-07:
Hi, Its runs in ESP32, I will update the library now
Regards
Joao

JoaoLopesF, 2017-09-07:
HI, I update it for the ESP32.
Please give me feedback

Regards

Joao

closed 2017-09-07 by JoaoLopesF

bospre, 2017-09-08:
Good Morning,
ok, thank you.
I will test, but need some time.

bospre, 2017-09-10:
Hello,
I have been testing and it seems to work.
Unfortunately my ESP32 triggers a "brownout" when attached via USB to the computer.
Sometimes it boots, most times not.
So I could not get the mDNS working.
For my project I need SPIFFS and cannot upload data to the SPIFFS.
I now ordered two more ESP32, to see if this is an issue with my ESP32. And I will wait until full SPIFFS integration is finished.