JoaoLopesF/RemoteDebug #17: Guidance using across multiple cpp files

lawrence-jeff, 2017-09-08:
Sorry for opening an issue for a question - not sure how else to ask.
I have a project where the functionality is split across several cpp files (and an .ino file shell that calls functions from them). They are full of serial debug statements which I would like to replace with your library. (I imagine not a unique scenario)

Could you provide some basic guidance for this scenario and where to do the includes and where to declare the object - I generally work with single file ino files so am not having much luck - all my attempts result in compiler errors. I believe it requires some combination of includes and extern declaration but can't get it working.

JoaoLopesF, 2017-09-08:
Hi, I too have projects with some cpp and h files
Works with cpp class or not.

To do RemoteDebug in each cpp, please do this:

----- INO ----

void setup() {

   // Your code .....

    test_SetRemoteDebug(Debug);

    // Or if it is a class

    test.setRemoteDebug(Debug);

}

----- CPP (not class) ----
#include "RemoteDebug.h"
RemoteDebug *_Debug;

void test_SetRemoteDebug(RemoteDebug *Debug) {
      _Debug = Debug;
}

void test1() {

   uint8_t a = 2 + 2;
   
   // Debug
   if(_Debug->isActive(_Debug->VERBOSE)) {
       _Debug->printf("test1: a-> %d\n", a);
    }
}

-----

Its clear now ?

Regards

Joao

closed 2017-09-08 by JoaoLopesF

lawrence-jeff, 2017-09-08:
Thank you for the reply, I very much appreciate the help as I use your library in another project and really like it, but still not having much luck in this case - I'm sure it's something I am doing based on my lack of familiarity with the ++ side of C, I am working on a fork so am stuck with this structure. Here is what I tried. Taking your approach I added the line 
test_SetRemoteDebug(Debug);
to the setup loop in the ino file, and added the other pieces to one of the .cpp files in my project (tweaked it with a single debug line). I didn't add any include or object instantiation in the ino file (since you didn't mention this) When compiling it errors on the ino file saying 'Debug' was not declared in this scope. I thought maybe I also needed to do the object creation in the ino so then tried adding 
RemoteDebug Debug;
So then upon compile the error changed to 

ogMainArduino:42: error: 'test_SetRemoteDebug' was not declared in this scope

But still no luck - tried a few other things but it doesn't like any of them.

If it makes it easier to see -the project link is below. For my testing/attempt I was trying to add a debug line to espconnect.cpp (and once I had a working example extend that to main.cpp and opengarage.cpp)

https://github.com/lawrence-jeff/OpenGarage-Firmware/tree/master/OpenGarage

Any pointers?

svenove, 2018-09-13:
I'm struggling with the same thing.
INO:
```
RemoteDebug Debug;
HanReader hanReader;
setup() {
  Debug.begin("AMS2MQTT", Debug.INFO);   
  hanReader.setRemoteDebug(Debug);
}
```

HanReader.cpp:
```
#include <RemoteDebug.h>
#include "HanReader.h"

static RemoteDebug *_Debug; // static -> private - for only this file

HanReader::HanReader()
{

}

void HanReader::setRemoteDebug(RemoteDebug *Debug) {
  _Debug = Debug;
}
```

HanReader.h:
```
class HanReader
{
public:
  void setRemoteDebug(RemoteDebug *Debug);
}
```

Error when compiling:
```
AmsToMqttBridge.ino: In function 'void setup()':

AmsToMqttBridge:75: error: no matching function for call to 'HanReader::setRemoteDebug(RemoteDebug&)'

     hanReader.setRemoteDebug(Debug);

                                   ^

AmsToMqttBridge.ino:75:35: note: candidate is:

In file included from AmsToMqttBridge.ino:13:0:

libraries\HanReader/HanReader.h:21:7: note: void HanReader::setRemoteDebug(RemoteDebug*)

  void setRemoteDebug(RemoteDebug *Debug);

       ^

libraries\HanReader/HanReader.h:21:7: note:   no known conversion for argument 1 from 'RemoteDebug' to 'RemoteDebug*'

exit status 1
no matching function for call to 'HanReader::setRemoteDebug(RemoteDebug&)'
```

It's probably just a silly error, but I'm unable to figure it out... :(