JoaoLopesF/RemoteDebug #18: Call remote debug with a single statement

gmag11, 2017-09-11:
Using `if ( Debug.isActive( .. ) ) ` is a bit annoying if you have lots of debug messages.

Using a function like 
``` Arduino
Debug.printf( debugLevel , "Some string with some format place holders %d %s", some, variables );
```
code would be cleaner.

it could be implemented as
``` Arduino
size_t RemoteDebug::printf (uint8_t debugLevel, const char* format , ...) {
    if (isActive (debugLevel)) {
        va_list argptr;
        va_start (argptr, format);
        printf (format, argptr);
        va_end (argptr);
    } 
}
```
Do you agree?

JoaoLopesF, 2017-11-24:
Hi, is a good suggestion. I make any tests with it and publish it soon

Thanks a lot

Joao

JoaoLopesF, 2017-11-24:
Hi

I have now released a new version with something similar to your idea.

The RemoteDebug should be with isActive because this has a smaller overhead.

But now we have shortcuts:

DEBUG ("This is any any (always showed) - var% d \ n", var);
DEBUG_V ("This is a message of debug level VERBOSE \ n");
DEBUG_D ("This is a message of debug level DEBUG \ n");
DEBUG_I ("This is a message of debug level INFO \ n");
DEBUG_W ("This is a message of debug level WARNING \ n");
DEBUG_E ("This is a message of debug level ERROR \ n");

I hope you enjoy

thankful
Joao

closed 2017-11-24 by JoaoLopesF