JoaoLopesF/RemoteDebug #24: Way of setting Telnet port

PjotrekSE, 2018-04-17:
Hi there, and thanx for this nice addition. Worx like a charm.
For reasons of obfuscating, I want to set the Telnet port to something other than 23. I designed a solution for this, see included files. I also changed the word 'setted' into 'set' in several places, hope you don't mind.
Keep up the good work!
PjoptrekSE

[RemoteDebug.cpp.txt](https://github.com/JoaoLopesF/RemoteDebug/files/1919294/RemoteDebug.cpp.txt)
[RemoteDebug.h.txt](https://github.com/JoaoLopesF/RemoteDebug/files/1919295/RemoteDebug.h.txt)

JoaoLopesF, 2018-05-19:
Hi, thanks for your suggestion,
I will do this soon

Regards,

Joao

JoaoLopesF, 2018-08-17:
I apreciate your suggestion, but is more simple, modify the port in RemoteDebug.h
I change the "setted"s, it will in next version
Thanks a lot

closed 2018-08-17 by JoaoLopesF

JoaoLopesF, 2018-08-17:
In the next version, I can modify for project:

#define TELNET_PORT 1000
#include "RemoteDebug.h"

Regards