JoaoLopesF/RemoteDebug #28: Can't compile sample for ESP32

EvertDekker, 2018-07-14:
Can't compile sample for ESP32, will give the following error:
C:\Users\Evert\Documents\Arduino\libraries\ESP8266-RemoteDebug-Telnet-master\Samples\RemoteDebug_Basic\RemoteDebug_Basic.ino:23:71: fatal error: ESP8266WiFi.h: No such file or directory

Don't get it wifi is working fine with mqtt and webserver.

stritti, 2018-07-22:
ESP8266WiFi  is just for ESP8266. You should change to a ESP32-compatible WiFi client.

JoaoLopesF, 2018-08-06:
Hi, sorry to delay time of this response

Same code is for ESP32 and ESP8266, so I use precompiler directives 
Please you use that IDE ?

Regards

Joao

EvertDekker, 2018-08-06:
Hi, 

I have changed somethings in RemoteDebug.cpp

```
#ifdef ESP32
  #include <WiFi.h>
#else
 #include <ESP8266WiFi.h>          
#endif
```

```
// Reset
        #ifdef ESP32
             ESP.restart();
        #else
             ESP.reset();
        #endif
```

Now it works.
Sorry, i'm not that good with Git else I had forked the solution.

I'm using 
Arduino Ide 1.8.5 with Esp8266 v2.3.0 and Esp32 V1.0

Btw, thanks for your great library

EvertDekker, 2018-08-17:
ESP.reset is what you used in your file.
To not break existing code I added the pre-compile directives.

ESP.restart works indeed on both cpu and ESP.reset only on the ESP8266
But it's not the same:

> ESP.reset() is a hard reset and can leave some of the registers in the old state which can lead to problems, its more or less like the reset button on the PC.
ESP.restart() tells the SDK to reboot, so its a more clean reboot, use this one if possible. 

However, ESP.restart() is the best and will do fine for both.

I will try the 1.3.0 version, thanks