JoaoLopesF/RemoteDebug #30: No debug messages displayed when using esp8266 boards package

Tiogaplanet, 2018-08-10:
I just upgraded my @esp8266 [core package](https://github.com/esp8266/Arduino)  to 2.4.2 and found that debug messages of any level no longer display to telnet.  

The introductory help message displays and responses to input commands still display too but no messages from my sketches are displayed.  This affects the example sketches included in the RemoteDebug library too.  For example, the code:

```
      DEBUG_V("* This is a message of debug level VERBOSE\n");
      DEBUG_D("* This is a message of debug level DEBUG\n");
      DEBUG_I("* This is a message of debug level INFO\n");
      DEBUG_W("* This is a message of debug level WARNING\n");
      DEBUG_E("* This is a message of debug level ERROR\n");
```

Will not display to telnet.  I tested this by editing the example code with my access point ssid and password only, no other changes and doing a compile on first the 2.4.1 version of esp8266 core then with the 2.4.2 core.  While both versions compile with no errors at all, debug messages are displayed with verison 2.4.1 but not with 2.4.2.

Tiogaplanet, 2018-08-10:
Did some quick experiments.  In `size_t RemoteDebug::write(uint8_t character)` currently no printing to telnet is done until `_bufferPrint.length() == BUFFER_PRINT` where BUFFER_PRINT is 150.  When bufferPrint.length() is 150, a string of "(d)" followed by 147 asterisks is printed to telnet.  The next time bufferPrint.length() gets to 150, 150 asterisks are printed to telnet.

However, if I change the test to `_bufferPrint.length() <= BUFFER_PRINT` then the correct output of:

```
* This is a message of debug level VERBOSE
* This is a message of debug level DEBUG
* This is a message of debug level INFO
* This is a message of debug level WARNING
* This is a message of debug level ERROR
```

is written to telnet but of course no actual buffering is done.

It appears that the 150 asterisks are the first asterisk of every debug message.  So it looks like only the first character of each debug message is getting stored to the buffer.

Tiogaplanet, 2018-08-10:
Disabling `CLIENT_BUFFERING` doesn't fix the problem. However changing the test to `(_bufferPrint.length() <= BUFFER_PRINT)`  makes all error messages print to telnet just as they should but only one character at a time.

Tiogaplanet, 2018-08-11:
Created a separate branch in my fork and removed all buffering and tried to do some code clean-up for consistency in comments and format.

Not sure if you're looking for this kind of contribution but if you are OK with it, I can continue cleaning up the examples and submit a PR once everything is cleaned up and uniform.

[My branch commit](https://github.com/Tiogaplanet/RemoteDebug/commit/3d3301753f85090bc476757a730fcc7d7fcacfac).

goosst, 2018-08-12:
>  makes all error messages print to telnet just as they should but only one character at a time

I seem to have the same issue  (only one character at the same time). It's not clear for me from this conversation if you had a fix for this?

Tiogaplanet, 2018-08-13:
I "fixed" the problem in that I changed `(_bufferPrint.length() == BUFFER_PRINT)` to `(_bufferPrint.length() <= BUFFER_PRINT)` which will allow RemoteDebug to resume sending debug messages to telnet.  That's the only change that needs to be made to get it working again.  Unfortunately, I don't know what is actually causing RemoteDebug to fail with version 2.4.2 in the first place.  

My commit that I linked in my earlier post includes the comparison fix and also attempts to do some significant code clean-up and impose some uniformity in how comments are used.

JoaoLopesF, 2018-08-17:
HI @Tiogaplanet 

With this change: 
(_bufferPrint.length() <= BUFFER_PRINT)
the buffering not works more, this is important to reduce overheads of RemoteDebug.
I just download the latest version of Arduino ESP8266 (stable)
And the problem appears ...
I research and found the bug: now need a second virtual write from Print.h
I make corrections and the version 1.3.0 is arrived.
Please try it (not need disable the buffering)

Regards

Joao

Tiogaplanet, 2018-08-18:
Appears to be working except for one problem.  Adding the hash to `#__VA_ARGS__` causes the user's debug message to be in two sets of double quotes.  The output of RemoteDebug_Basic.ino looks like this:

```
"(d) * This is a message of debug level DEBUG\n""(i) * This is a message of debug level INFO\n"
```

Removing the hash so that your rdebug statements contain `__VA_ARGS__` produces the expected output:

```
(d) * This is a message of debug level DEBUG
(i) * This is a message of debug level INFO
```

JoaoLopesF, 2018-08-18:
Hi @Tiogaplanet 

Ops, this is a test to resolve the issue
I just commit  new version without this (1.3.1)
Thanks for your help, You has been fast to detect this

Regards
Joao

Tiogaplanet, 2018-08-18:
You’ve written a great library and I’m using it in a [project](https://github.com/Tiogaplanet/MiP_ESP8266_Library) of my own. I’d like to help out where I can.

closed 2018-08-18 by Tiogaplanet