JoaoLopesF/RemoteDebug #31: Compile Error in Platform.io IDE

stritti, 2018-09-01:
I have updated to current version 1.4 getting now compile errors. E.g.
```
.piolibdeps\RemoteDebug_ID1266/RemoteDebug.h:180:39: error: expected primary-expression before '...' token
#define rdebug(fmt, ...) rdebugA(fmt, ...)
^
```
See also travis-build: https://travis-ci.org/stritti/smart-swimming-pool

stritti, 2018-09-01:
I think  line https://github.com/JoaoLopesF/RemoteDebug/blob/eb80cf8fc5c5b8b792a09a577ca3ec9b342f63a4/RemoteDebug.h#L180

should be changed to:

```#define rdebug(fmt, ...) rdebugA(fmt, ##__VA_ARGS__)```

JoaoLopesF, 2018-09-03:
Hi, @stritti 

Yes, Your solution is correct 
I just updating the library to 1.5.2

JoaoLopesF, 2018-09-04:
Hi @stritti ,

Please verify now, I update library.json

stritti, 2018-09-04:
now it works, thank you very much @JoaoLopesF !

closed 2018-09-04 by stritti