JoaoLopesF/RemoteDebug #33: Serial output not working

jeroenst, 2018-09-03:
When using this code:
  Debug.setSerialEnabled(true);
  Debug.begin(esp_hostname);
  Debug.setResetCmdEnabled(true);
  Debug.setCallBackProjectCmds(&processCmdRemoteDebug);
  DEBUG("\n");
  DEBUG("ESP8266 started...\n");
  Serial.println("\n\nHello\n");

The ESP8266 started (and everyting with the debug macro after this code) is not printed.
The Hello is printed.

What's wrong?

jeroenst, 2018-09-03:
Bug is caused by _PasswordOk on line 445

JoaoLopesF, 2018-09-04:
Hi @jeroenst 

Yes, It is correct.
I made a version 1.5.3, only use this if for passwords

jeroenst, 2018-09-05:
I see your fix. But I think it would be better that serial debugging always works if enabled, also when a password is used on the telnet interface (which is great actually!).

jeroenst, 2018-09-05:
Also it would be better to set password through a function so it can be changed while running the program on the esp imo

jeroenst, 2018-09-05:
Tested your fix, it works when no password is defined. When a password is defined, the RemoteDebug lib has building errors.

JoaoLopesF, 2018-10-05:
@jeroenst , sorry by delay.
Please send the error

jeroenst, 2018-10-06:
/home/jeroen/Arduino/libraries/RemoteDebug/RemoteDebug.cpp: In member function 'boolean RemoteDebug::isActive(uint8_t)':
/home/jeroen/Arduino/libraries/RemoteDebug/RemoteDebug.cpp:454:17: error: '_PasswordOk' was not declared in this scope
  boolean ret = (_PasswordOk &&
                 ^
/home/jeroen/Arduino/libraries/RemoteDebug/RemoteDebug.cpp: In member function 'void RemoteDebug::showHelp()':
/home/jeroen/Arduino/libraries/RemoteDebug/RemoteDebug.cpp:744:15: error: '_PasswordAttempt' was not declared in this scope
   help.concat(_PasswordAttempt);
               ^
/home/jeroen/Arduino/libraries/RemoteDebug/RemoteDebug.cpp: In member function 'void RemoteDebug::processCommand()':
/home/jeroen/Arduino/libraries/RemoteDebug/RemoteDebug.cpp:1139:44: error: a function-definition is not allowed here before '{' token
 void RemoteDebug::setFilter(String filter) {
                                            ^
/home/jeroen/Arduino/libraries/RemoteDebug/RemoteDebug.cpp:1150:33: error: a function-definition is not allowed here before '{' token
 void RemoteDebug::setNoFilter() {
                                 ^
/home/jeroen/Arduino/libraries/RemoteDebug/RemoteDebug.cpp:1161:77: error: a function-definition is not allowed here before '{' token
 String RemoteDebug::formatNumber(uint32_t value, uint8_t size, char insert) {
                                                                             ^
/home/jeroen/Arduino/libraries/RemoteDebug/RemoteDebug.cpp:1199:1: error: expected '}' at end of input
 }
 ^
exit status 1
Error compiling for board Generic ESP8285 Module.

jeroenst, 2018-10-06:
_passwordOk in .h is _PasswordOk in .cpp
_passwordAttempt in .h is _PasswordAttempt in .cpp
and there is a } missing on line 891 of RemoteDebug.cpp

closed 2019-01-15 by JoaoLopesF

jeroenst, 2019-07-04:
Works with version 3.0.5