JoaoLopesF/RemoteDebug #36: Short way debugging

foorschtbar, 2018-10-19:
Hey, short way debugging without any paramter doenst work. The docu says rdebugV("text") but this is wrong...

JoaoLopesF, 2018-10-19:
Hi, please send example how You is using it

foorschtbar, 2018-11-13:
rdebugVln("This is a println"); doesnt work for me.

JoaoLopesF, 2018-11-13:
Hi, @foorschtbar 

I tried this and works well.

Please give this informations:

Version of Arduino IDE (or another IDE):
Operational system:

Board:

Regards

Joao Lopes

foorschtbar, 2018-11-14:
- Arduino 1.8.6 @ Windows 10 x64
- KiTTY as Telnet Client

*** Remote debug - over telnet - for ESP8266 (NodeMCU) - version 1.5.6
* Host name: roombaesp IP:192.168.xxxx Mac address:18:FE:34:Dxxxx
* Free Heap RAM: 23528
* ESP SDK version: 2.2.1(cfd48f3)
******************************************************

JoaoLopesF, 2018-11-14:
Hi, @foorschtbar 

Thanks for information. I tested now the lib in Esp8266 and works well.
I do it in Arduino 1.8.7. 

Please set Arduino to show verbose in compilation and send me all output.

Regards

Joao Lopes

foorschtbar, 2018-11-15:
Hi @JoaoLopesF,

here the log, telnet output and a test sketch.

Regards,
Fabian

[telnet.txt](https://github.com/JoaoLopesF/RemoteDebug/files/2586556/telnet.txt)
[verboselog.txt](https://github.com/JoaoLopesF/RemoteDebug/files/2586557/verboselog.txt)
[Sketch.zip](https://github.com/JoaoLopesF/RemoteDebug/files/2586560/Sketch.zip)

JoaoLopesF, 2018-11-15:
HI, @foorschtbar 

I just download the files.

And now I understand.

The debugs #1 and #2 are of level verbose.
And the default level of this library is Debug,
due it, this debugs not is showed.

Before access page,  you need give "v" command to set level to verbose,
and this debug will showed.
A tip, if You use sufix "ln", not need put "\n"in message.

The #3 and #4 are of level always, due it is showed always

Please try it and close this issue.

Regards

Joao Lopes

foorschtbar, 2018-11-15:
Hi @JoaoLopesF,

thank you for your support. I thought that only rdebugV[ln] could be called without parameters (looked like this in your example). That's why I used it. rdebugA[ln]("Hello") works too. My mistake. I just had to look into the source code.

The \n was only for testing. Thank you!

Regards,
Fabian

closed 2018-11-15 by foorschtbar