JoaoLopesF/RemoteDebug #37: If telnet password is set the timeout due to inactivity is always set to 1 min

achuchev, 2018-11-20:
Looking at the code below the telnet connection will be closed after 1 min of inactivity even when MAX_TIME_INACTIVE is set to different value.
  
`#ifdef MAX_TIME_INACTIVE

		// Inactivity - close connection if not received commands from user in telnet
		// For reduce overheads

		uint32_t maxTime = MAX_TIME_INACTIVE; // Normal

		if (_password != "") { // Request password - 18/08/08
			maxTime = 60000; // One minute to password
		}

		if ((millis() - _lastTimeCommand) > maxTime) {
			TelnetClient.println("* Closing session by inactivity");
			TelnetClient.stop();
			_connected = false;
			_silence = false;
		}
#endif`

JoaoLopesF, 2018-12-03:
Hi @achuchev 

Please download lastest version with this fixed

Thanks for your feedback

achuchev, 2018-12-05:
@JoaoLopesF Yes, it works!

closed 2018-12-05 by achuchev