JoaoLopesF/RemoteDebug #38: Getting a debug in startup

EnoraNedelec, 2018-12-02:
Hi, 

I am trying to print some values from the startup. 
I tried to do it by doing this:
1) using your basic example
2) right before the end of startup (below the long list of Serial.println()), I add these lines:
```
delay(15000);  // this gives me time to connect Putty, 
rdebugDln("startup done");
delay(1000); 
Debug.handle();
delay(5000); // instead of the  yield();
```

But in Putty, I dont see the `startup done`, I only see the debug from the `loop`. 
In the docs you mention that `Debug.handle();` should be in loop. But is there a way to also use it in `startup`.

JoaoLopesF, 2018-12-03:
HI @MagTun 

To debug setup code, I recomend you enable serialoutput:
Debug.setSerialEnabled(true);
And use serial monitor to see the messages, not need extra delays
And when done, you can disable serial output, just comment the command setSerialEnabled.

For me, It is a better solution, that I use when need debug setup codes.
Server and high delays is not good ...

Regards

João

MagTun, 2018-12-09:
Thanks for your reply and sorry for the delays. So it means that if I can't connect my nodemcu to my computer to get the serial monitor, I won't be able to get a debug from `start`?

closed 2019-01-15 by JoaoLopesF