JoaoLopesF/RemoteDebug #39: Compile errors with wemos d1 mini

zxxxh, 2019-01-10:
Compiling for Wemos D1 Mini using Arduino IDE. I receive the following compiler errors.

/Users/zach/Documents/Arduino/libraries/RemoteDebug/RemoteDebug.cpp: In member function 'void RemoteDebug::begin(String, uint16_t, uint8_t)':
/Users/zach/Documents/Arduino/libraries/RemoteDebug/RemoteDebug.cpp:104:25: error: no matching function for call to 'WiFiServer::begin(uint16_t&)'
  TelnetServer.begin(port);
                         ^
/Users/zach/Documents/Arduino/libraries/RemoteDebug/RemoteDebug.cpp:104:25: note: candidate is:
In file included from /Users/zach/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:40:0,
                 from /Users/zach/Documents/Arduino/libraries/RemoteDebug/RemoteDebug.cpp:65:
/Users/zach/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiServer.h:53:8: note: virtual void WiFiServer::begin()
   void begin();
        ^
/Users/zach/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiServer.h:53:8: note:   candidate expects 0 arguments, 1 provided
exit status 1
Error compiling for board WeMos D1 R2 & mini.

tilutza, 2019-01-10:
In the version 2.3 and up the begin function does not accepts a parameter. I fixed in the branch

https://github.com/JoaoLopesF/RemoteDebug/pull/40

JoaoLopesF, 2019-01-15:
I just merged your adjustment, thanks a lot

closed 2019-01-15 by JoaoLopesF