JoaoLopesF/RemoteDebug #44: no display using debugA

jetpax, 2019-03-09:
First, thanks for a wonderful utility

I found debugA doesn't display

changed line 247 of RemoteDebug.h from 

`#define debugA(fmt, ...) rdebugVln(fmt, ##__VA_ARGS__)`

to

`#define debugA(fmt, ...) rdebugAln(fmt, ##__VA_ARGS__)`

and seems to work

JoaoLopesF, 2019-03-09:
Hi @jetpax, thanks to found this bug.
I will correct the code
Regards

JoaoLopesF, 2019-03-11:
Hi @jetpax , this is fixed in lastest version (3.0.0)
Thanks a lot for your colaboration

jetpax, 2019-03-11:
Thanks Joao, look forward to the web app

Funnily enough I am using your RemoteDebug to debug an issue in _my_ webapp, (resets after 6 hours or so) but think this may be the [asyncwebserver](https://github.com/me-no-dev/ESPAsyncWebServer) that I am using

Actually this shows what maybe a bug in RemoteDebug but I will submit that on another ticket

cmidgley, 2019-03-23:
This does not appear to be fixed - as of 3.0.4 it is still "V" instead of "A".  Tnx!

JoaoLopesF, 2019-03-23:
Hi @cmidgley and @jetpax ,
My mistake, I changed in incorrect directory and not published to repo
Please download the latest version.
Thanks for your feedback

karol-brejna-i, 2023-05-24:
It works. My version is 3.0.4