JoaoLopesF/RemoteDebug #45: debug messages repeated and ANSI escape showing

jetpax, 2019-03-11:
I'm seeing what looks like a buffer overflow when sending debug messages, that displays the ANSI escape message )see image)

I don't see them ever when RemoteDebug sends 'Project commands", eg u command in image,
but as you can see when debugV for example sends data from my program, sometimes the previous response is repeated with part of the ANSI escape  sequence missing and hence appears as text.

[image]

JoaoLopesF, 2019-03-12:
Hi @jetpax 

Any answers:
- That telnet client you use and system operational ?
- That version of RemoteDebug ?
- It occurs always or sometimes ?

I not see it before, so do I make a simple sketch to reproduce it ? (to isolate another logic)

JoaoLopesF, 2019-03-12:
Please add some your logic to example advanced of RemoteDebug to reproduce it

JoaoLopesF, 2019-03-12:
I do some debugI, but not lost the escape on ansi:
[image]

jetpax, 2019-03-12:
MacOS 10.14.2, homebrew telnet client v.60, occurs about 50% of the time.

effect is always last message is repeated, but with first char missing (ie escape)

interestingly, I don't have a function called operator() as shown in the above, so not sure where that comes from....  maybe because debug is in a callback function (of asyncwebserver, the server.on method, see below)

ie
```
	server.on("/api/vehicle", HTTP_PATCH, [=](AsyncWebServerRequest *request){
		unsigned int code = 200;

		if(request->hasParam("name", true) &&  request->hasParam("value", true) ){
		
		AsyncWebParameter* p = request->getParam("name", true);
		const char *name = p->value().c_str();

		p = request->getParam("value", true);
		const char *value = p->value().c_str();
		
		debugI("webapi Received %s = %s", name, value);

		if (strcmp(name, "speed") == 0) {
			vehicle_speed = strtol(value, NULL, 10);
		} else if (strcmp(name, "rpm") == 0) {
			vehicle_rpm = strtol(value, NULL, 10);
		} else if (strcmp(name, "throttle") == 0) {
			vehicle_throttle = strtof(value, NULL);
		} else if (strcmp(name, "vin") == 0) {
			strncpy(vin, value, 17);
		}
		} else {
		debugW("webapi Invalid data received !");
		code = 400;
		}
		request->send(code);
	});

```

I don't see this when use debug in 'regular' code, and in this case the function name is correct eg

[image]

JoaoLopesF, 2019-03-12:
@jetpax , I am use MacOS and telnet 60 :-)
Any strange behavior !
One workarround:
 at end of function write, before the client print,
you can check if it starts with '[' and show a message (to we known that) and preffix the escape.

JoaoLopesF, 2019-03-12:
So we knowed that it is due a firwware or telnet client issue

JoaoLopesF, 2019-03-13:
You need help on workaround?

tablatronix, 2020-09-30:
is this the same issue?, osx terminal, seems like a bug.
I think I can reproduce it as above toggling.

[image]