JoaoLopesF/RemoteDebug #46: Disable WS is not working

22MarioZ, 2019-03-18:
**Describe the bug**
The switch `#define WEBSOCKET_DISABLED true` is not working. I'm trying to use RemoteDebug, but I'm not able to disable WebSocket. I need to disable websocket as I'm using it in my code.

I try also the Advance Example and the behavior is the same.

**To Reproduce**
Steps to reproduce the behavior:
1.  Compile the "RemoteDebug_Advanced" example with `#define WEBSOCKET_DISABLED true`
2. Connect to ESP with "http://joaolopesf.net/remotedebugapp/"
3. If the WS ic disabled I should not be able to connect with application.

**Expected behavior**
If the switch `#define WEBSOCKET_DISABLED true` is enabled, the connection with remotedebugapp should not be possible.

**Arduino Information:**
 - OS: MacOSX,
 - IDE: Atom + PlatformIO
 - IDE: latest
 - Board: ESP32

JoaoLopesF, 2019-03-18:
Hi @22MarioZ 

Your lib for sockets is https://github.com/Links2004/arduinoWebSockets ?
If isit, you can use the web app to debug,
just put in your project :
#define USE_LIB_WEBSOCKET true

About disable, I will do some tests and return

JoaoLopesF, 2019-03-18:
bug confirmed

JoaoLopesF, 2019-03-18:
Workarround until find the solution:

do it in library source RemoteDebug.h: @line 43

JoaoLopesF, 2019-03-18:
And dowload the new version 3.0.3, with corrections for WS disabled

JoaoLopesF, 2019-03-19:
Hi @22MarioZ ,
I saw how it occurs:
the problem is the library is compiled before the main code.
So the "#define" not works at all.
The solution is do it in RemoteDebug.h
I will change the examples to avoid this

Thanks to add this issue.

22MarioZ, 2019-03-19:
Hi @JoaoLopesF 

Thank you for the help. Yes. I'm using same WebSocket library. I will follow your suggestion and disable  WS in the RemoteDebug.h.

JoaoLopesF, 2019-03-19:
Hi @22MarioZ ,

I just published a new version, with warnings about use of #define for RemoteDebug.
Thanks to add this issue.

closed 2019-03-19 by JoaoLopesF