JoaoLopesF/RemoteDebug #47: Break execution until connected

user1969903, 2019-03-29:
Would it be possible to add a function which will pause the execution of all code until some connects to the debugger?

It would be useful in situations where, for example, you have some code that is executed during setup, before you have the chance to connect.

JoaoLopesF, 2019-03-31:
HI @user1969903 

You can use this code, after Debug init codes :

while (!(Debug.isConnected())) {
   delay(250);
}

You can use the setSerialEnabled to see debugs in setup.

user1969903, 2019-04-23:
Thanks for the info.
I actually used the custom function capability of the library to write my own 'break' function and called it during setup but it is good to know that this functionality exists out of the box. Is this documented anywhere? 
I was not able to find it prior to asking this question. If it isn't, perhaps you could include a list of all exposed methods on the main page of this git.

JoaoLopesF, 2019-05-09:
Hi, sorry by delay on response

I will see it soon, thanks for this feedback

Regards

einsteinx2, 2019-07-24:
I ended up using a small modification in my sketch so that in case I never connect the sketch will still run. I chose a 20 second delay to ensure Wifi had time to fully connect (it seems to take somewhere between 10 and 15 seconds on my network using a static IP, but you can tweak it to suite your needs).

Here's the code:
```
unsigned long launchMillis = millis();
while (!Debug.isConnected()) {
    Debug.handle();
    if (millis() - launchMillis > 20000) break; // Allow 20 seconds to connect before starting
    delay(100);
}
```