JoaoLopesF/RemoteDebug #48: WiFiClientSecure.h not found on platform espressif32 (platformio)

majodi, 2019-03-31:
**Describe the bug**
WiFiClientSecure.h not found error when using espressif32 platform with PlatformIO

**To Reproduce**
Just use remotedebug in a project on PlatformIO with espressif32

**Expected behavior**
The code should work also under these conditions

**Additional context**
I used the WEBSOCKET_DISABLED switch but as PlatformIO is compiling the /utility directory I also had to add an entry to library.json:

  "build": {
    "srcFilter": "+<*> -<utility/>"
  },

Now my project works again but of course this is not ideal... Better to fix the code for espressif32 use.

JoaoLopesF, 2019-04-06:
Hi @majodi

Sorry for delay, I not have time to test it in platformio, 
But I saw a lot of issues in it about the include. 
When I install platformio, I will work in a solution

Thanks for this feedback

majodi, 2019-04-06:
No problem. I think for other potential casual PIO users like me it would be good to know that one can use:

(platformio.ini)

build_flags =
    -DWEBSOCKET_DISABLED=true

to set the switch. But because PIO still performs a compile on the /utility dir you have to avoid this somehow. I added an entry to library.json. But I don't know what the best way would be to dynamically set this option to exclude the /utility dir completely when needed.
Also if the code there would be suitable for the espressif32 platform it wouldn't matter. So I think it would be best to have conditional code that compiles with espressif32, then the exclude would not be needed.

ISWlehre, 2019-04-10:
For some Reason I do get a similar error with Wificlientsecure.h in Arduino IDE while compiling. My programming-skills are not advanced enough to figure out the reason or fix the problem myself.
Maybe I'm also just doing something wrong.