JoaoLopesF/RemoteDebug #50: Including WebSockets files breaks projects using the library

mairas, 2019-05-07:
If you try to build a project that uses the WebSockets library, the files in the RemoteDebug project cause linking errors (multiple definitions of class methods).

I'm writing a library that is using the RemoteDebug library. My library is using WebSockets natively. When building, I get a lot of these linker errors:

```
.pioenvs/d1_mini/libdf9/libWebSockets_ID549.a(WebSockets.cpp.o): In function `WebSockets::write(WSclient_t*, unsigned char*, unsigned int)':
WebSockets.cpp:(.text._ZN10WebSockets5writeEP10WSclient_tPhj+0x10): multiple definition of `WebSockets::write(WSclient_t*, unsigned char*, unsigned int)'
.pioenvs/d1_mini/lib5ee/libRemoteDebug_ID1266.a(WebSockets.cpp.o):WebSockets.cpp:(.text._ZN10WebSockets5writeEP10WSclient_tPhj+0x10): first defined here
.pioenvs/d1_mini/libdf9/libWebSockets_ID549.a(WebSockets.cpp.o): In function `WebSockets::sendFrame(WSclient_t*, WSopcode_t, unsigned char*, unsigned int, bool, bool)':
WebSockets.cpp:(.text._ZN10WebSockets9sendFrameEP10WSclient_t10WSopcode_tPhjbb+0x24): multiple definition of `WebSockets::sendFrame(WSclient_t*, WSopcode_t, unsigned char*, unsigned int, bool, bool)'
.pioenvs/d1_mini/lib5ee/libRemoteDebug_ID1266.a(WebSockets.cpp.o):WebSockets.cpp:(.text._ZN10WebSockets9sendFrameEP10WSclient_t10WSopcode_tPhjbb+0x24): first defined here
.pioenvs/d1_mini/libdf9/libWebSockets_ID549.a(WebSockets.cpp.o): In function `WebSockets::clientDisconnect(WSclient_t*, unsigned short, char*, unsigned int)':
WebSockets.cpp:(.text._ZN10WebSockets16clientDisconnectEP10WSclient_ttPcj+0x4): multiple definition of `WebSockets::clientDisconnect(WSclient_t*, unsigned short, char*, unsigned int)'
.pioenvs/d1_mini/lib5ee/libRemoteDebug_ID1266.a(WebSockets.cpp.o):WebSockets.cpp:(.text._ZN10WebSockets16clientDisconnectEP10WSclient_ttPcj+0x4): first defined here
.pioenvs/d1_mini/libdf9/libWebSockets_ID549.a(WebSockets.cpp.o): In function `WebSockets::handleWebsocketPayloadCb(WSclient_t*, bool, unsigned char*)':
WebSockets.cpp:(.text._ZN10WebSockets24handleWebsocketPayloadCbEP10WSclient_tbPh+0x18): multiple definition of `WebSockets::handleWebsocketPayloadCb(WSclient_t*, bool, unsigned char*)'
...
```

Please consider either pulling the WebSockets in as an external dependency (probably works only in PlatformIO) or renaming the WebSockets classes e.g.: WebSockets -> RDWebSockets etc.

JoaoLopesF, 2019-05-09:
Hi, Sorry by delay on response.

Please set this in your project:

// Internally, the RemoteDebug uses a local copy of the arduinoWebSockets library (https://github.com/Links2004/arduinoWebSockets)
// Due it not in Arduino Library Manager
// If your project already use this library,
// Uncomment the line below, to do it:
#define USE_LIB_WEBSOCKET true

This config exist for it

Regards

mairas, 2019-05-26:
Hi, sorry for the delay on my behalf as well. Unfortunately, defining `USE_LIB_WEBSOCKET` is not enough: it does pull the correct includes but linking still fails because all cpp files are still pulled in. If you added the ifdefs around the `src/utility/WebSockets*.cpp` files, I guess it might work.

sparky3387, 2019-08-15:
I also had the issue, according to this you must define the websockets disabled = true, cannot be set in your main and must be set in the remotedebugcfg.h,  (Tested and confirmed by me at least)

[#define directive ignored on IDE 1.8.2](https://forum.arduino.cc/index.php?topic=480702.0)