JoaoLopesF/RemoteDebug #52: Remote Debug für Arduino Mega/Uno/nao

bospre, 2019-05-12:
Hello,
I am using your Remote Debug on several ESPxx Projects.
But I also have some projects with Arduino Mega and Ethernet Shield.
Would it not be possible to use it also with the Arduino?
Boris

JoaoLopesF, 2019-05-13:
HI, bospre
I not have the ethernet shield, but You can try use this with mega,
if wifi libraries is compatible

Regards

Joao

Please notice me if you can sucess on it.

bospre, 2019-05-14:
Hello Joao,
thanks for reply.
I tried, but it does not compile.
At first I had to remove the lines which point to ESPxx im remotedebug.h (line 99:113)
But it still does not compile:
C:\Users\boris\AppData\Local\Temp\arduino_modified_sketch_318151\RemoteDebug_Basic.ino:185:4: note: in expansion of macro 'debugE'
    debugE("* This is a message of debug level ERROR");
    ^
exit status 1
And there I am ending with my knowledge
As far as I know, the network libraries are very similar, so it shouldn't be a problem to change the remotedebug, so it will run also on arduino.
If you want, I can send you an ethernet shield.
Boris

peros550, 2020-06-06:
Any news with Arduino Mega? Is it supported?