JoaoLopesF/RemoteDebug #53: namespace breaks macros?

ccoenen, 2019-06-16:
**Describe the bug**
When inside a namespace, I can't use the `debugI`, `debugV` ... shorthands.

**Code Example**

This prints out the same thing, once via tha long form and once with the `debugI` macro.

```cpp
namespace Example {
	void test(const char * prefix) {
		#ifndef DEBUG_DISABLED
		if (Debug.isActive(Debug.INFO)) {
			Debug.printf(prefix, "hi");
		}
		#endif
		debugI(prefix, "hi");
	}
};
```

Now call `Example::testWorks("this says %s");`.

**Expected behavior**
Both ways of printing a debug line should work.

**Current behaviour**

Here's the compiler output:

```
some-file: In function 'void Example::test(const char*)':
some-file:29:10: error: expected ')' before 'prefix'
   debugI(prefix, "hi");
          ^

src/RemoteDebug.h:146:84: note: in definition of macro 'rdebugI'
   #define rdebugI(fmt, ...) if (Debug.isActive(Debug.INFO))   Debug.printf("(%s) " fmt, __func__, ##__VA_ARGS__)
                                                                                    ^

src/RemoteDebug.h:195:26: note: in expansion of macro 'rdebugIln'
 #define debugI(fmt, ...) rdebugIln(fmt, ##__VA_ARGS__)
                          ^

some-file:29:3: note: in expansion of macro 'debugI'
   debugI(prefix, "hi");
   ^
```

**Arduino Information:**
 - OS: Windows
 - IDE Arduino: IDE
 - IDE Version: 1.8.9
 - Board: ESP8266

JoaoLopesF, 2019-06-19:
Hi, In your code I saw that if.. Debug.printf appears compile, try use this instead debugI macro
Regards

ccoenen, 2019-06-19:
Yes. the long form does work. It is what i'm using in the meantime. Is there any possibility to get the shorthands working? I am not too familiar with the way macros are expanded, so I'm not sure if this is possible at all.