JoaoLopesF/RemoteDebug #59: The domain joaolopesf.net expired on 10/10/2019 and is pending renewal or deletion

skleeschulte, 2019-10-17:
**Describe the bug**
The domain joaolopesf.net expired on 10/10/2019 and is pending renewal or deletion.

**To Reproduce**
Navigate browser to http://joaolopesf.net/remotedebugapp.

**Expected behavior**
Opening http://joaolopesf.net/remotedebugapp shows remote debug web-app.

jeroenst, 2019-11-24:
It's pretty quiet on jaolopesf, he hasn't commited anything since 19 june 2019....

oori, 2020-03-01:
Here's a temporary link you can use:  http://remotedebugapp.pluggable.biz

I can't promise how long it'll be online.  I recommend you host the RemoteDebugApp locally:  just [download](https://github.com/JoaoLopesF/RemoteDebugApp/archive/master.zip) the app, and [serve it](https://github.com/lwsjs/local-web-server) locally.

jeroenst, 2020-03-02:
I think that @JoaoLopesF is not active anymore :-(

jhubbardsf, 2020-08-10:
I think we should accept that Joao might be taking an extended vacation... What's the best way to figure out which fork should be the next project fork after a project manager disappears like this?

jeroenst, 2020-08-10:
That's a good question. I have a fork with latetst pull request and did some debugging in the code, but I don't have any ambition to solve other bugs for this project, altough I will implement pull requests from this project when requested.

I think someone who is more commited than me in coding is a better choice.

jhubbardsf, 2020-08-13:
Same here. I might end up doing some work to the RemoteDebugger repo but I don't feel like I have the bandwidth to do much work to this repo and that one. I do think this is the best remote debugging software for Arduino right now and it would be a shame for the work put into it not to be continued.