JoaoLopesF/RemoteDebug #64: Ability to use flash strings in debug messages

kostuch, 2020-02-13:
**Is your feature request related to a problem? Please describe.**

All messages processed by debug are stored in RAM memory.
So in effect, observing/debugging application causes huge difference in RAM utilisation.

**Describe the solution you'd like**

Something like:
`debugI(F("some message stored in flash"));`

Besides: excellent library.

jeroenst, 2020-03-02:
I doubt if the messages are stored in RAM, maybe for just a little time until it is processed, but is that really an issue?

kostuch, 2020-03-02:
As far as I know, all commands like debugI("Debug message") or Serial.print("something to print") by default put string data to the RAM.
When I compile my app with many debug inserts, I can see that RAM utilization grows rapidly.
If there is a huge amount of RAM available, then there is no issue, but sometime we hit the limit and additional consumption by debug is a problem.

karol-brejna-i, 2023-05-27:
In RemoteDebugCfg.h I see:
```
// Enable Flash variables support - F()
// Used internally in SerialDebug and in public API
// If is a low memory board, like AVR, all strings in SerialDebug is using flash memory
// If have RAM memory, this is more fast than flash
// #define DEBUG_USE_FLASH_F true
```

It probably does what @kostuch wants? @kostuch , have you tried this option?

jeroenst, 2024-01-26:
You can use this:

```
if (Debug.isActive(Debug.INFO)) Debug.printf(cF("example"));
```

With cf and this macro's:

```
#define _gFmtBufSize 100+1

// ERB - Force format strings and string constants into FLASH Memory
#define sF(x) String( F(x) )
         // Used as an F() when being used as the first Element
         // of a Multi-Element Expression
char _gFmtBuf[_gFmtBufSize];
         // Buffer, Used with cF() to store constants in program space (FLASH)
#define cF(x) strncpy_P(_gFmtBuf, (PGM_P)PSTR(x), sizeof(_gFmtBuf)) 
// Used with printf() for the char format string
```