JoaoLopesF/RemoteDebug #77: Websockets library error and Ethernet access

MassiPi, 2021-06-08:
**Describe the bug**
- with recent ESP8266 cores, the embedded websockets library gives error compiling. The library should be updated
- with 3.0.0 esp8266 core, and new embedded ethernet functionalities, RemoteDebug is not working (no telnet connection can be made)

**To Reproduce**
Steps to reproduce the behavior:
1. build a project with 3.0.0 esp8266 core to get the websockets error
2.build a project with 3.0.0 esp8266 and W5500 ethernet shield. no connection can be made

**Expected behavior**
- the library should build also with new core versions
- the library should work also over ethernet

**Arduino Information:**
 - OS: Windows
 - IDE Platformio
 - Board ESP8266 with 3.0.0 core

blaskovicz, 2021-07-07:
Is this the websockets error you are referring to? I'm unable to build on the d1 mini and want to confirm if it's similar.

```
C:\Users\zacau\Documents\Arduino\libraries\RemoteDebug\src\utility\WebSocketsClient.cpp: In member function 'void WebSocketsClient::connectedCb()':
C:\Users\zacau\Documents\Arduino\libraries\RemoteDebug\src\utility\WebSocketsClient.cpp:704:26: error: 'class BearSSL::WiFiClientSecure' has no member named 'verify'
  704 |         if(!_client.ssl->verify(_fingerprint.c_str(), _host.c_str())) {
|                          ^~~~~~```

Related to #78

MassiPi, 2021-07-07:
yes, it is.
It's due to the fact that remotedebug contains also an old version of the webockets library.
In my case, i do not use RD over websockets so i just commented out that part and disabled websockets, but i assume this is a design error to include an external library in this way.
If you use websockets with RD, i think there is a way to use the updated websockets library, included externally (something like "use external websockets" in remotedebug.h, i can't check now)

Btw, it seems RD works correctly also with the new ethernet lwip library, very good!