JoaoLopesF/RemoteDebug #80: error: 'class BearSSL::WiFiClientSecure' has no member named 'verify'

eaglefred, 2021-08-13:
**Describe the bug**
Error when compiling my source code for ESP8266

**Arduino Information:**
 - OS: Linux
 - IDE ArduinoIDE 1.8.15
 - Board ESP8266 GenericModule

**Additional context**
Here is the compilation error:
/home/fred/Arduino/libraries/RemoteDebug/src/utility/WebSocketsClient.cpp: In member function 'void WebSocketsClient::connectedCb()':
/home/fred/Arduino/libraries/RemoteDebug/src/utility/WebSocketsClient.cpp:704:26: error: 'class BearSSL::WiFiClientSecure' has no member named 'verify'
  704 |         if(!_client.ssl->verify(_fingerprint.c_str(), _host.c_str())) {
      |                          ^~~~~~

robin-s-007, 2021-08-22:
Hi eaglefred,

I have run into the same problems in the last couple of days.
I have used this library with success on the ESP32 in the past, by now I wanted to use it on the ESP8266, and got the same problems. 
I am using platformIO and am using the arduino framework, before I started my project (last weekend), I updated all the platform libraries.
What I have come across during the last couple of days is that there seems to have been a change in the arduino framework sometime ago, and the local websocket libraries that are included in the RemoteDebug library do not match anymore. 

I have been able to solve it in the following manner:
- install the latest WebSocket library directly from the source: https://github.com/Links2004/arduinoWebSockets/tree/master/src
  This is the same library as was originally included into RemoteDebug, but only the newest version.
- Enable:  #define USE_LIB_WEBSOCKET true
   in the RemoteDebugCfg.h 

Then clean and rebuild, it should then build again. 

This worked for me to get it compling again, but have not tested the code yet. However I expect that it will work now.

Best Regards,

Robin

eaglefred, 2021-08-24:
Hi Robin,
Thank you very much for the trick, i have tried it but it doesn't work for me. I have an error when compiling:

exit status 1
'RemoteDebug' does not name a type

Without the #include "RemoteDebug" and all its refering code, my source code is compiling, so i have to look for another way to debug it.
Tank you for your message.
Take care.

Fred

Urs-Eppenberger, 2021-08-24:
Just to let you know, I'm running in exactly the same problem.
And since I'm not a programmer, just barely able to patch together some code, I can't fix this on my own.

Urs-Eppenberger, 2021-08-24:
I downgraded to version 2.1.2
Now it works again.
I'm happy using telnet as debug possibility. I do not need the web functionality.
(since I'm working with sonoff devices which are connected to 220V life power, the debug functionality over WiFi us a big improvement on security for me)  ;-)

foorschtbar, 2021-11-28:
Some problem here. Downgrade to 2.1.2. Looks like the repo here will no longer be maintained :(

thanek, 2022-01-25:
If you want to use the new Arduino library, you have to exclude some files from compilation process. In order to do that in platformio you can use the `extra_scripts` platformio feature and point to a script which will filter compiled files and effectively exclude those in the `utility` folder. You can check the example in my repository here:
https://github.com/thanek/sharp-fu-y30-mqtt-esp/blob/master/platformio.ini
and the script:
https://github.com/thanek/sharp-fu-y30-mqtt-esp/blob/master/utils/filter_src.py

This, combined with the `-DUSE_LIB_WEBSOCKET=1` build flag should do the job.
Hope that helps.

scottchiefbaker, 2023-02-16:
Last commit on this repo was 2019. Unfortunately it looks like this project has been abandoned :(

I was able to "fix" this issue by downgrading to v2.1.2 as mentioned. This allowed me to compile for a NodeMCU board just fine. Hopefully that will help someone else.

DrStein99, 2023-03-13:
I confirm, downgrading to v2.1.2 fixed compiler errors (which had me up all night tracing down).  Having trouble getting the debugger to reply to my prompts, via telnet.  Trying to figure out what I am doing wrong, and how many other functions this does (or not) interfere with.

scottchiefbaker, 2023-03-13:
I've started work on a similar tool to RemoteDebug. It's still a work in progress, but if you're interested in testing something similar check out [RIX](https://github.com/scottchiefbaker/ESP-RIX).