JoaoLopesF/RemoteDebug #87: SHA resource no longer correct for ESP32

TheMadTexan42, 2022-09-26:
Can no longer build on ESP32 due to a change in the ESP32 SHA libraries.

hwcrypto/sha.h no longer exists.

The workaround seems to be to replace
#include <hwcrypto/sha.h>
in
\WebSockets.cpp (line42)
by
#include <esp32/sha.h>

(see https://github.com/espressif/esp-idf/issues/3245)

However this is a recent change so ultimately it depends on the toolchain version you have in place for the ESP

alex-eri, 2022-09-29:
#warning esp32/sha.h is deprecated, please use sha_parallel_engine.h instead

JonLevin25, 2022-12-28:
Is there a way to `#define <hwcrypto/sha.h> <esp32/sha.h>` from user code, to avoid changing library?

The preprocessor doesn't seem to like it

```log
In file included from src/main.cpp:5:
src/remote_debug.h:31:9: error: macro names must be identifiers
 #define <hwcrypto/sha.h> <esp32/sha.h>
```

quazardous, 2024-04-05:
here is a good patch

https://github.com/Links2004/arduinoWebSockets/issues/738#issuecomment-1043356325

```cpp
  #if ESP_IDF_VERSION_MAJOR >= 4
    #if ( ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(1, 0, 6) )
      #include "sha/sha_parallel_engine.h"
    #else
      #include <esp32/sha.h>
    #endif  
  #else
    #include <hwcrypto/sha.h>
  #endif
```

quazardous, 2024-04-05:
crap just use https://github.com/karol-brejna-i/RemoteDebug (RemoteDebug2)