JoaoLopesF/RemoteDebug #90: Drop-in replacement for this library

karol-brejna-i, 2023-05-29:
I found this library very useful for my purposes. Combining OTA and remote debugging capabilities made things easier for my development efforts. The library is also popular among other Arduino/ESP developers.

Unfortunately, it seems like the library is not maintained anymore. The last commit was made in 2019. There are "hanging" pull requests and unattended issues.

The biggest problem with the library is that it simply doesn't compile in its current form. 

I've gone through the issues and forks (there are almost 120 of them!). They are focusing mostly on the build problem. It is located here: https://github.com/karol-brejna-i/RemoteDebug.

Because I am actively using the library, I decided to fork it and make some fixes.

So if you are interested in including the functionality, you could use the fork I prepared. It is a drop-in replacement. You only need to change the dependency from `joaolopesf/RemoteDebug@^3.0.5` to `karol-brejna-i/RemoteDebug@^4.0.0`.

The changes I made include:
- Removed ArduinoWebsockets from the sources and used the one from the library manager
- Included changes from the following PRs (not included in the original library ATTOW):
  - included changes from https://github.com/JoaoLopesF/RemoteDebug/pull/73
  - included changes from https://github.com/JoaoLopesF/RemoteDebug/pull/56
- Removed unused code, cleaned up the sources
- Make WebSockets (with the dedicated web app) work again
- Provided some more documentation, examples
```

Of course, I'd be glad to contribute back to the original repo if the author is interested in that...

karol-brejna-i, 2023-05-30:
Also, I'd appreciate some feedback if the library works for your bords. I use only a small number of different ESP32 devices...

mskressin, 2023-08-30:
Do you see any reason why your branch would not work or could be modified to work with the new Arduino Uno R4 WIFI board? I have not tried it yet, but since the Arduino Uno R4 WIFI uses the ESP32-S3 chip for WIFI, I thought I would ask since you know the code and it's dependencies.

karol-brejna-i, 2023-08-31:
> Do you see any reason why your branch would not work or could be modified to work with the new Arduino Uno R4 WIFI board? I have not tried it yet, but since the Arduino Uno R4 WIFI uses the ESP32-S3 chip for WIFI, I thought I would ask since you know the code and it's dependencies.

@mskressin I haven't had any chance to work with the board you mentioned.  It should work. 
Please, try it out and let me know. If there are problems, we should be able to resolve them quickly.
If it works, let me know, too ;-)

I remember there were some problems with the [arduino-esp32](https://github.com/espressif/arduino-esp32) library itself (some missing declarations for ESP32-S3), but it looks like they are solved already: https://github.com/espressif/arduino-esp32/pull/8187#event-10229722511.

Anyway, let's continue the discussion in the [new repo](https://github.com/karol-brejna-i/RemoteDebug).